probe-rs/hs-probe #1: DAPLink firmware

flit, 2020-06-18:
Very cool design! You should consider using the [DAPLink](https://github.com/ARMmbed/DAPLink) firmware. We don't yet support the F723, but it's been on the requested and we'd happily accept contributions. Other work is ongoing by a partner to support the H747.

korken89, 2020-06-19:
Hi, thanks for the kind words!

The hardware design was made with the main focus of being a reference USB 2.0 High-Speed probe for probe-rs. This does not impact support for other firmware, but unfortunately I do not have time to put currently.

If anyone in your development community is interested to get a probe for developing firmware, they can contact me and I'll send a probe to the person. :)
Plus if there is feedback on missing features or ideas they are most welcome!

Have a good day!

flit, 2020-06-19:
Understood, I'm very familiar with lack of time! Either me or a colleague will be in touch about a board soon. 😄

korken89, 2020-06-22:
Do so, it would be awesome to see more open hardware high-speed probes!