probe-rs/hs-probe #5: 5V Out

dbrgn, 2020-09-04:
The JLink allows powering the target device through [pin 19 of the 20-pin connector](https://www.segger.com/products/debug-probes/j-link/technology/interface-description/) or [pin 7 of the 10-pin connector (opposite of TDI)](https://github.com/gfroerli/hardware-v2/blob/master/datasheets/ARM-JTAG-20-10_latest-schematic.pdf) (at least on the Olimex adapter, this pin is often used for keying as well). To enable this feature, you need to send the "power on" command to the debug probe first. You can chose between "power on" (temporarily enable 5V/USB power out until the probe is unplugged) or "power on perm" (permanently enable power on that pin).

It would be nice if the hs-probe would support this as well! I use that feature extensively, it saves a lot of time if I don't need a separate power supply.

korken89, 2020-09-04:
In general I see no issue with this, but the STDC14 connector has GND on the KEY pin.
One could add a dual MOSFET to switch between GND or 5V.

Right now we have followed the STLinkV3 implementation and supply 5V on pin 1 of the STDC14 connector.

korken89, 2020-11-28:
I fixed this for v1.3 :)

PR #13 (2020-11-28): Added FET to control KEY pin
With the moving of the tag connect we can now fit the FET that was omitted before.

Fixes #5
closed 2020-11-28 by Yatekii