probe-rs/hs-probe #7: How to get a hs-probe?

Sympatron, 2020-10-20:
Is there any plan to produce and sell these?

The layout doesn't quite lend itself to hand soldering, which makes low volume production (especially one piece) not really feasible.

I think this project is a great idea, because it enables to use a full debugging stack written in pure open source Rust, which would be awesome!

korken89, 2020-10-20:
Hi!

Indeed there are plans! We have gotten quotes from a few manufacturers for a run of 100 probes and we are just hashing out the final design details that has come up during testing.
Exactly how it will be sold is not 100% clear right now, but it will be done somehow. :)

I will soon start a "I want one" list where people can sign up and with quantity as I will need to lay out with a (for me significant) sum of money until most probes are sold, and I'll make an order when we have enough interested.
If it does not get to enough I will most likely manufacture them myself (up to maybe 20 pcs) to increase awareness and see how demand goes from there. :)

Thanks for showing interest!

BR Emil

Sympatron, 2020-10-20:
Where do you plan to announce updates to this?

korken89, 2020-10-20:
I have planned to use the following channels:

- Probe-rs Matrix Channel
- Rust Embedded Matrix Channel
- This and the firmware repository
- Rust Embedded newsletter

korken89, 2020-10-28:
The list is up, here is a link for the one looking:
https://docs.google.com/forms/d/e/1FAIpQLScj-8rHTeytxxHBYu0chEmUiG58iWNTc8dejDOi9G5oWHEf3w/viewform

werediver, 2022-08-10:
The [shop](https://shop.probe.rs/) is currently down (reports 403 Forbidden). Are there any plans to bring it up again?

Yatekii, 2022-08-10:
The current semiconductor market is in terrible shape still, unfortunately. Which is why we cannot manufacture more of the probe, which is also why I took down the link on the page.

closed 2022-08-10 by Yatekii