probe-rs/hs-probe #9: What about an additional standard 10-pin Cortex-M debug header?

sirhcel, 2020-10-23:
Thank you for your awesome work! Today we discussed in [#probe-rs:matrix.org](https://matrix.to/#/#probe-rs:matrix.org) that having a standard 10-pin Cortex-M debug header would be a great feature for future releases if space is no concern.

[image]

A 14 pin cable does not fit in all cases and not eveyone has 1,27 mm cables and connectors around. These connectors could be placed directly besides each other as only one cable should be connected at a time.

sirhcel, 2020-10-23:
Wrong button ... :-|

Sympatron, 2020-10-27:
I would vote for a 14-10 cable (maybe optional) since they are easy to manufacture. Even a DIY kit (14-pin connector, 10-pin connector and a 10 pin cable) would be sufficient in my opinion.

korken89, 2020-10-27:
Hi, I have tested making a 14-10 pin cable and it went great.
So for the order I will add so one can check if they want a 14-14 cable and/or a 14-10 cable.
For the ones using tag-connect cables there is already compatible cables.

Having 2 connectors on the board is no problem technically, but the connector used on the board is 1/5 the price of the total BOM.
So I am not very keen on having 2. :)

sirhcel, 2020-10-28:
That's good news! Having the option to get an appropriate cable or its parts along with the probe solves this issue for me.

> Having 2 connectors on the board is no problem technically, but the connector used on the board is 1/5 the price of the total BOM.
> So I am not very keen on having 2. :)

I see your point and agree.

Please close this issue at your discretion.

korken89, 2020-10-28:
Glad to hear it! Closing for now.

closed 2020-10-28 by korken89