probe-rs/hs-probe #12: USB TTL UART Capability

dbrgn, 2020-11-24:
I'm not sure if this is possible over the shared USB port, but: When debugging a device with UART output, it would be fantastic if the hs-probe could also be used as a serial adapter via USB.

Is it possible that a USB device can be used both as a programmer/debugger, and as a separate serial adapter?

Right now I always need to plug in two cables into my computer: The JLink and a generic USB serial adapter... It would be great if those could be combined. (Yes, I'm aware of RTT, but some projects need to use UART...)

Disasm, 2020-11-24:
hs-probe for sure can be used in this way, but we don't have a firmware support yet. But this is definitely something to be implemented!

dbrgn, 2020-11-24:
Ah, awesome! :tada: 

What about the hardware? The SWD pinout doesn't have serial RX/TX. But on this rendering...

[image]

...there are two RX/TX pins on the right side. Could those be used, given firmware support?

Maybe a dedicated 3-pin header (RX/TX/GND) could be added in a future version.

Disasm, 2020-11-24:
RX/TX signals are also available on the STDC14 connector. The ones on the board edge are the same signals.

dbrgn, 2020-11-24:
> RX/TX signals are also available on the STDC14 connector.

Ah, interesting! So far I've mostly used the 10-pin SWD connector in my projects, but that might be a reason to switch to the 14-pin connector.

> Actually, it's already present in firmware and available to your host system, but all the data comes directly into the void.

@Yatekii I'm very much looking forward to the first hardware prototypes, I might give this a try :slightly_smiling_face: