probe-rs/hs-probe #16: Allow for different VDD_Target

allexoll, 2021-02-27:
In the current version the probe allows for 3v3 and 5v, it might be interesting to add 1v8 functionality. 

If we want it to exclusively work with 1v8, that could be done by just switching the MIC5353-3.3YMT for a MIC5353-1.8YMT and the STM32F723 supports it.
But if we need a range of functionality (most of segger's [probes](https://www.segger.com/products/debug-probes/j-link/models/model-overview/) do 1.2v -> 5V) , that might require something more complex.

allexoll, 2021-05-21:
I think what is usually done  is using VDD_TARGET to power the target side of a level translator.

korken89, 2021-05-21:
The main issue with level translators is that they need to be bidirectional, and with that often come with problematic special cases.
So personally I'm not in favor of a level translator approach.

What one could do though is to let the MCU set the resistor divider and control it's voltage in the range 1.8 - 3.3v.

allexoll, 2021-07-13:
at least for SWD that's doable ie: NXP's LPC55S16-EVK

[image]

What kind of edge case were you thinking about? 

having a resistor div for the LDO would be doable, but you're limited to the vdd range of the probe's MCU (ie: no 5V target)

i'm unsure how segger or ST do it

diggit, 2022-04-11:
This is the only reason keeping me from using hs-probe. Are then any plans?
I am also willing t help with HW development if there is agreed solution.

allexoll, 2022-04-11:
@korken89 has been working on a new [rp2040 probe](https://github.com/probe-rs/rusty-probe), which includes adjustable signaling, you should look that up

closed 2022-04-11 by allexoll